bbulkow/FastLED-idf #1: Undefined reference to 'pinMode' with ESP32/APA102

JadElClemens, 2020-03-02:
The example project compiles fine with the default example configuration, but if I change the `FastLED.addLeds` call to `FastLED.addLeds<APA102, 12, 14, BGR>(leds, NUM_LEDS)` (just for example) I get a linker error `components/FastLED-idf/platforms/esp/32/fastpin_esp32.h:20: undefined reference to 'pinMode'`.

This may just be a bug in FastLED, but I figured I'd report it here since I'm not using vanilla FastLED.

bbulkow, 2020-03-08:
I see. It appears that APA102 uses SPI only, and won't work with RMT like the WS8211s. Thus all of the LED types which require SPI are non-functional. I'm checking to see if it's something simple to get the SPI code properly compiling, but I don't have LEDs of those sorts to test with. Pull request welcome!

bbulkow, 2020-03-09:
Fixed, it compiles now. pinMode is an Arduino function that doesn't exist in esp-idf, which means pulling in more of the hal.

I don't have any 4-wire LEDs around the house. If you do, please try and let me know if it works.

Interestingly, a 4-wire pin can't use the RMT interface, because the two wires have to be in sync,  which means the speed of this system is far lower. No parallel channels.

I have updated the README in this regard.

The upstream FastLED code does not support hardware SPI either.

closed 2020-03-09 by bbulkow