bbulkow/FastLED-idf #4: Linking issues

drewandre, 2020-03-15:
Hi, I'm so grateful you took the time to port this library! I think it'll be well worth the effort given the popularity of esp32 and fastled.

I wanted to let you know of couple issues I had trying to build this example repository.
1. I didn't see any clear instructions in the readme on how to run. I'm assuming I can just run `make`, but do I need to include the arduino library as well? Some clear instructions here would be helpful for future users, even if its just git clone and make.
2. `esp32-hal.h` was not found in two files - FastLED.h and platforms/esp/32/clockless_rmt_esp32.h. I changed these to hal/esp32-hal.h which resolved that issue.
3. I'm currently stuck on this one - I'm getting linking errors related to any timing functions like `millis`, `delay`, and `micros` (see output below). Would you recommend I include Arduino in my components folder?

I'm using ESP-IDF version 4.1 and xtensa-esp32-elf-gcc toolchain version 5.2.0.
```
LD build/FastLED-idf.elf
/Users/drewandre/esp/FastLED-idf/build/main/libmain.a(main.o):(.literal._Z16blinkLeds_simplePv+0x10): undefined reference to `delay'
/Users/drewandre/esp/FastLED-idf/build/main/libmain.a(main.o): In function `CFastLED::show()':
/Users/drewandre/esp/FastLED-idf/components/FastLED-idf/controller.h:171: undefined reference to `delay'
/Users/drewandre/esp/FastLED-idf/build/FastLED-idf/libFastLED-idf.a(FastLED.o):(.literal._ZN8CFastLED8countFPSEi+0x8): undefined reference to `millis'
/Users/drewandre/esp/FastLED-idf/build/FastLED-idf/libFastLED-idf.a(FastLED.o):(.literal._ZN8CFastLED4showEh+0x4): undefined reference to `micros'
/Users/drewandre/esp/FastLED-idf/build/FastLED-idf/libFastLED-idf.a(FastLED.o): In function `CFastLED::countFPS(int)':
/Users/drewandre/esp/FastLED-idf/components/FastLED-idf/FastLED.cpp:236: undefined reference to `millis'
/Users/drewandre/esp/FastLED-idf/components/FastLED-idf/FastLED.cpp:236: undefined reference to `millis'
/Users/drewandre/esp/FastLED-idf/build/FastLED-idf/libFastLED-idf.a(FastLED.o): In function `CFastLED::show(unsigned char)':
/Users/drewandre/esp/FastLED-idf/components/FastLED-idf/FastLED.cpp:236: undefined reference to `micros'
/Users/drewandre/esp/FastLED-idf/components/FastLED-idf/FastLED.cpp:236: undefined reference to `micros'
collect2: error: ld returned 1 exit status
make: *** [/Users/drewandre/esp/FastLED-idf/build/FastLED-idf.elf] Error 1
```

dant14, 2020-03-20:
Firstly - commendable job on this port - I tried it with the SPI APA102 LEDs from sparkfun and I can confirm that the basic functionality works perfectly at a first glance.

I had various issues ( linking errors such as the above one as well I think ) but I just did
`git checkout release/v4.0` in the esp-idf folder and everything worked as it should.

bbulkow, 2020-03-28:
First, I have not tested on 4.1, only on the 4.0 release branch. I am hoping ESP-IDF is stable enough that it works on 4.x but have not tested.

Second, you shouldn't have to include Arduino, it is not intended. It is intended that the project builds when you download it, and then you can do the work of moving the component into your own project. The example doesn't include arduino.

Does the example make without link errors, before you copy the component over? That's the first step, before including it in your own project.

Would it be possible to try the rel/4.0 branch, I will see if I can get some time to try ether 4.1/rel or master?

bbulkow, 2020-04-07:
Hi, I'm going to close this issue as I've updated the README to be clear the project is an ESP-IDF project and must be built with the ESP-IDF system, and provided links to that environment.

closed 2020-04-07 by bbulkow