bbulkow/FastLED-idf #5: idf.py build fails

geofholbrook, 2020-05-07:
This version of FastLED appears to be what I need ... but it doesn't seem to build with the latest version of esp-idf. (`v4.2-dev-1206-g741960d5c`)

`idf.py build` gives the following read out (slightly abridged)

```Executing action: all (aliases: build)
...
[7/367] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_bus_lock.c.obj
FAILED: esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_bus_lock.c.obj 
...
~/Data/programming/triwave/esp/esp-idf/components/driver/spi_bus_lock.c: In function 'spi_bus_lock_init_main_dev':
~/Data/programming/triwave/esp/esp-idf/components/driver/spi_bus_lock.c:822:41: error: implicit declaration of function 'xSemaphoreCreateBinaryStatic'; did you mean 'xSemaphoreCreateBinary'? [-Werror=implicit-function-declaration]
     g_spi_lock_main_flash_dev->semphr = xSemaphoreCreateBinaryStatic(&main_flash_semphr);
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                         xSemaphoreCreateBinary
~/Data/programming/triwave/esp/esp-idf/components/driver/spi_bus_lock.c:822:39: warning: assignment to 'SemaphoreHandle_t' {aka 'void *'} from 'int' makes pointer from integer without a cast [-Wint-conversion]
     g_spi_lock_main_flash_dev->semphr = xSemaphoreCreateBinaryStatic(&main_flash_semphr);
                                       ^
cc1: some warnings being treated as errors
[8/367] Linking CXX static library esp-idf/nghttp/libnghttp.a
ninja: build stopped: subcommand failed.
ninja failed with exit code 1```

bbulkow, 2020-05-10:
Both of those error sare in esp-idf, and not in FastLED, however, I think I know your problem, and am moving the project to 4.2 as we speak.

I have found in order to move from 4.0 to 4.2, it is necessary to do a fresh clone, and follow the steps of install as well as export.

However, this particular problem is having a menuconfig which is out of sync with the version of ESP-IDF. In particular, the OS functionality of creating Static semaphores is covered by a capability that must be enabled via menuconfig, and in 4.0, that was off by default, now it appears to be required on, because you can see that in the SPI code in the driver, it's required.

I will be checking in changes to work with 4.2 shortly, just checking a few things in the gpio definitions that changed.

geofholbrook, 2020-05-11:
@bbulkow thank you! 

I have had success so far with some other projects, and the examples in the esp-idf repo itself  tend to compile ok. I did a fresh clone as you suggested, get a different error on build. I could hold off until I see a new commit here since you mention you're moving to 4.2, but I'll include the readout in case it's useful:

```[5/12] Building C object esp-idf/FastLED-idf/CMakeFiles/__idf_FastLED-idf.dir/hal/esp32-hal-gpio.c.obj
FAILED: esp-idf/FastLED-idf/CMakeFiles/__idf_FastLED-idf.dir/hal/esp32-hal-gpio.c.obj 

...

In file included from /Users/geof/Data/programming/triwave/esp/esp-idf/components/soc/soc/esp32/../include/soc/gpio_periph.h:20,
                 from /Users/geof/Data/programming/triwave/esp/esp-idf/components/driver/include/driver/gpio.h:22,
                 from /Users/geof/Data/programming/triwave/esp/esp-idf/components/driver/include/driver/rtc_io.h:20,
                 from ../components/FastLED-idf/hal/esp32-hal-gpio.c:84:
/Users/geof/Data/programming/triwave/esp/esp-idf/components/soc/soc/esp32/include/soc/gpio_caps.h:23: warning: "GPIO_PIN_COUNT" redefined
 #define GPIO_PIN_COUNT          (40)
 
../components/FastLED-idf/hal/esp32-hal-gpio.c:30: note: this is the location of the previous definition
 #define GPIO_PIN_COUNT 40
 
../components/FastLED-idf/hal/esp32-hal-gpio.c: In function '__pinMode':
../components/FastLED-idf/hal/esp32-hal-gpio.c:93:24: error: 'rtc_gpio_desc' undeclared (first use in this function); did you mean 'rtc_io_desc'?
     uint32_t rtc_reg = rtc_gpio_desc[pin].reg;
                        ^~~~~~~~~~~~~
                        rtc_io_desc
../components/FastLED-idf/hal/esp32-hal-gpio.c:93:24: note: each undeclared identifier is reported only once for each function it appears in
[6/12] Linking CXX static library esp-idf/libsodium/liblibsodium.a
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

bbulkow, 2020-06-19:
That I already solved. One adds the menuconfig option to support the 'legacy' use of the rtc_gpio_desc' structure. However, reading the esp-idf code, it's unclear why one doesn't just use the esp_gpio_desc structure. They're the same but one's easier to find.

bbulkow, 2020-06-28:
Fixed now.

closed 2020-06-28 by bbulkow