bbulkow/FastLED-idf #7: ESP-IDF driver doesn't work right

bbulkow, 2020-06-30:
In release versions 4.1, 4.2, and master, the system driver is enabled by default, because the code doesn't crash.

However, it also doesn't work. The colors are terribly wrong, so it seems the conversion going into ESP-IDF must be bad, or the bit rate is bad, or something.

First test will be to see if the build-in interrupt handler at least is working under the later releases just so something works, then to go after the actual flaw causing the driver to not work.

plebed, 2020-07-16:
Maybe the problem is caused by https://github.com/espressif/esp-idf/issues/5524

bbulkow, 2020-07-20:
Thanks for the pointer. This repo is now a fork of https://github.com/samguyer/FastLED , and thus all the code has changed, so needs to be retested. I don't think Sam's code was correct for the built in driver - I sent him a pull request.

I will keep an eye on 5524, thanks.

bbulkow, 2020-09-15:
Seems to work now.

closed 2020-09-15 by bbulkow