bbulkow/FastLED-idf #8: build failed after enable bluetooth

yawyd, 2020-06-30:
Hello

`idf.py build` works fine after I clone this awesome repo.
But it went wrong after I execute:
```bash
make menuconfig
// then, choose Components->Bluetooth-> and enable Bluetooth item
```
I got error message like this:
```bash
../components/FastLED-idf/hal/esp32-hal-misc.c:31:10: fatal error: esp_bt.h: No such file or directory
 #include "esp_bt.h"
          ^~~~~~~~~~
compilation terminated.
```

I'm using esp-idf version **v4.0.1** on **Ubuntu20.20**

for now, I did some _dirty_ by **comment the line 31 in file `components/FastLED-idf/hal/esp32-hal-misc.c`** and build pass.

Let me know I miss somethings?

bbulkow, 2020-07-24:
Ah. good catch. Let me take a look.

bbulkow, 2020-07-27:
Resolved. Please re-pull

closed 2020-07-27 by bbulkow