bbulkow/FastLED-idf #9: implicit declaration of function 'pinMatrixOutAttach'

Ivorforce, 2020-06-30:
I got further some steps, and now I'm getting a compile failure on various hal-files.

I traced back the steps and this function seems to be usually declared in `esp32-hal-matrix.h`, which is not included in this distribution.

I have to add I'm using platformio's Arduino / esp-idf special cocktail mix, and the arduino framework adds those header files itself too. Though it does not seem to be compiled along with this in this case.

bbulkow, 2020-07-24:
Since I don't use platformio, I can't reproduce this. Happy to take a pull request of your fix though.

bbulkow, 2020-07-27:
Rather interesting. The file that was annoying you isn't used, the hal files aren't for both of those --- the RMT system is direct and doesn't need an arduino "shim".

Thus, I've removed the files and references to them. I can only assume that PlatformIO is simply taking all the C files and compiling them, instead of looking in the CMakeLists.txt and trying to figure out what files are actually used.

Let me know if this is resolved.

closed 2020-07-27 by bbulkow