bbulkow/FastLED-idf #10: Thank you for your work

tonyp7, 2020-07-28:
Just a small message to say I appreciate the work you've done with this.

This past week I have really pulled my hair on controlling a WS2812 with esp-idf and RMT. Most notably I had issues where sometimes some LEDs were not getting lighted up properly and that happened at random --and that's with using a proper SN74LVC1T45 level translator. I turned to FastLED but it's a too deeply intertwined with Arduino code that I do not want to touch this with a 10 foot pole.

I figured there must be some issues at play here with multi-threading and other interrupts generating issues with the RMT routine, though I could never confirm that.

I fixed the issue by doing all the WS2812 processing in its own task with a queue/message pump. Since then I haven't had any issues. I think the 10ms of the FreeRTOS default tick time is enough get interference with the rest of the code, although again I am not 100% sure about this, it's my hypothesis.

[Code here](https://github.com/tonyp7/esp32-nixie-clock/blob/master/nixie_clock_code/main/ws2812.c#L104) if you wish to peek.

bbulkow, 2020-07-30:
Looks like you didn't use this library, according to your code? Did you have issues using this component, that I can look into?

tonyp7, 2020-07-31:
Not at all! I used a different codebase to control WS2812s because I had not found your repo yet when I started. I was just ranting about the same glitches you mention while using network and the lack of a good RGB led lib for the esp32 ;-)

tonyp7, 2020-07-31:
closing as to not pollute your issue tracker

closed 2020-07-31 by tonyp7