bbulkow/FastLED-idf #11: Build error "RMT_DEFAULT_CONFIG_TX" not declared.

jacobandersson86, 2020-08-24:
I am getting the build error as stated in snippet below. I am using esp-idf v4.0. Does anyone have any ideas what might be wrong here? I am trying to build the example, but I am getting the same problem when I am trying to incoperate FastLED-idf to my project. 
```
[2/8] Building CXX object esp-idf/FastLED-idf/CMakeFiles/__idf_FastLED-idf.dir/platforms/esp/32/clockless_rmt_esp32.cpp.obj
FAILED: esp-idf/FastLED-idf/CMakeFiles/__idf_FastLED-idf.dir/platforms/esp/32/clockless_rmt_esp32.cpp.obj 
/Users/jacobandersson/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++   -Iconfig -I../components/FastLED-idf -I../components/FastLED-idf/hal -I/Users/jacobandersson/esp/esp-idf/components/newlib/platform_include -I/Users/jacobandersson/esp/esp-idf/components/freertos/include -I/Users/jacobandersson/esp/esp-idf/components/heap/include -I/Users/jacobandersson/esp/esp-idf/components/log/include -I/Users/jacobandersson/esp/esp-idf/components/soc/esp32/include -I/Users/jacobandersson/esp/esp-idf/components/soc/include -I/Users/jacobandersson/esp/esp-idf/components/esp_rom/include -I/Users/jacobandersson/esp/esp-idf/components/esp_common/include -I/Users/jacobandersson/esp/esp-idf/components/xtensa/include -I/Users/jacobandersson/esp/esp-idf/components/xtensa/esp32/include -I/Users/jacobandersson/esp/esp-idf/components/esp32/include -I/Users/jacobandersson/esp/esp-idf/components/driver/include -I/Users/jacobandersson/esp/esp-idf/components/esp_ringbuf/include -I/Users/jacobandersson/esp/esp-idf/components/esp_event/include -I/Users/jacobandersson/esp/esp-idf/components/tcpip_adapter/include -I/Users/jacobandersson/esp/esp-idf/components/lwip/include/apps -I/Users/jacobandersson/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/jacobandersson/esp/esp-idf/components/lwip/lwip/src/include -I/Users/jacobandersson/esp/esp-idf/components/lwip/port/esp32/include -I/Users/jacobandersson/esp/esp-idf/components/lwip/port/esp32/include/arch -I/Users/jacobandersson/esp/esp-idf/components/vfs/include -I/Users/jacobandersson/esp/esp-idf/components/esp_wifi/include -I/Users/jacobandersson/esp/esp-idf/components/esp_wifi/esp32/include -I/Users/jacobandersson/esp/esp-idf/components/esp_eth/include -I/Users/jacobandersson/esp/esp-idf/components/efuse/include -I/Users/jacobandersson/esp/esp-idf/components/efuse/esp32/include -I/Users/jacobandersson/esp/esp-idf/components/app_trace/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu++11 -fno-exceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\"v4.0\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/FastLED-idf/CMakeFiles/__idf_FastLED-idf.dir/platforms/esp/32/clockless_rmt_esp32.cpp.obj -MF esp-idf/FastLED-idf/CMakeFiles/__idf_FastLED-idf.dir/platforms/esp/32/clockless_rmt_esp32.cpp.obj.d -o esp-idf/FastLED-idf/CMakeFiles/__idf_FastLED-idf.dir/platforms/esp/32/clockless_rmt_esp32.cpp.obj -c ../components/FastLED-idf/platforms/esp/32/clockless_rmt_esp32.cpp
../components/FastLED-idf/platforms/esp/32/clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::init()':
../components/FastLED-idf/platforms/esp/32/clockless_rmt_esp32.cpp:100:31: error: 'RMT_DEFAULT_CONFIG_TX' was not declared in this scope
         rmt_config_t rmt_tx = RMT_DEFAULT_CONFIG_TX((gpio_num_t)0, rmt_channel_t(i));
                               ^~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

jacobandersson86, 2020-08-24:
Reverted one commit to df6963230107d0a6fe38d6e981506245586b0fce and it does now compile the example.

bbulkow, 2020-09-07:
This is due to the annoyance that ESP-IDF does _not_ support semantic versioning. The initialization of the rmt_config_t is different from 4.0 and 4.1++. I had found a recipe to make it all work before.... let me see if I can find it again....

bbulkow, 2020-09-12:
Resolved, has now been tested with 4.0 and 4.2 and master.

closed 2020-09-12 by bbulkow