bbulkow/FastLED-idf #12: Adding an LED strip on GPIO33 fails

outlandnish, 2020-08-28:
First, thanks for creating this port for ESP-IDF!

I just switched from IDF 3.3 to IDF 4.1 and now whenever I add a Neopixel LED controller to GPIO33, I get the following error:

```
0x40088559: __pinMode at c:\users\nishanth\code\firmware-amp-idf\build/../components/FastLED-idf/hal/esp32-hal-gpio.c:131

0x400e3026: ClocklessController<33, 20, 50, 30, (EOrder)66, 0, false, 5>::init() at c:\users\nishanth\code\firmware-amp-idf\build/../components/FastLED-idf/platforms/esp/32/clockless_i2s_esp32.h:230

0x4010efbd: CFastLED::addLeds(CLEDController*, CRGB*, int, int) at c:\users\nishanth\code\firmware-amp-idf\build/../components/FastLED-idf/FastLED.cpp:39 (discriminator 4)
```

The code works fine for the other GPIO pins that I use (22, 26, and 14). This code worked fine for GPIO33 with IDF 3.3.

outlandnish, 2020-08-28:
This following section is not needed in `esp32-hal-gpio.c` if the `Support array `rtc_gpio_desc` for ESP32` option in `Component config > Driver configurations > RTCIO configuration` is enabled in `menuconfig`
```c++
#if ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(4,1,0)
#define RTC_IO_DESC (rtc_gpio_desc)
#else
#define RTC_IO_DESC (rtc_io_desc)
#endif
```
That said, even with that change, I kept getting the error above. Since I'm not using any of RTC functionality, I commented out this section and I no longer have the issue. That first `lock rtc` line is what causes the error but I'm not sure why that's happening.

```c++
if(rtc_reg) {
        //lock rtc
        ESP_REG(rtc_reg) = ESP_REG(rtc_reg) & ~(RTC_IO_DESC[pin].mux);
        if(mode & PULLUP) {
            ESP_REG(rtc_reg) = (ESP_REG(rtc_reg) | RTC_IO_DESC[pin].pullup) & ~(RTC_IO_DESC[pin].pulldown);
        } else if(mode & PULLDOWN) {
            ESP_REG(rtc_reg) = (ESP_REG(rtc_reg) | RTC_IO_DESC[pin].pulldown) & ~(RTC_IO_DESC[pin].pullup);
        } else {
            ESP_REG(rtc_reg) = ESP_REG(rtc_reg) & ~(RTC_IO_DESC[pin].pullup | RTC_IO_DESC[pin].pulldown);
        }
        //unlock rtc
    }
```

bbulkow, 2020-09-07:
Interesting, thanks. The fact that the IDF_VERSION is not _needed_ doesn't make me want to take it out, because then I have to check for existence. I'll look into it though.

I'll check into this, though, it seems just defining as PIN 33 shows the error, so it should be quick to sort out. Thanks.

bbulkow, 2020-09-12:
There was a code error around checking for RTC pins and doing the right thing, and the deprication of the old API in 4.1, and my poor understanding of the issue in my workaround. I believe I have re-coded correctly for the new API with changelist 81e041d2554fee36be9df2ff8e9303753e926536 , please let me know if the problem has not be resolved.

bbulkow, 2020-09-12:
Going to mark this resolved. Please retest and reopen if you have an issue.

closed 2020-09-12 by bbulkow

outlandnish, 2020-09-15:
@bbulkow sorry for the delay in responding. I just confirmed that everything works now, thanks!