bbulkow/FastLED-idf #14: APA102 LoadProhibited when Bluetooth is enabled and data pin 19 is used

drewandre, 2020-09-02:
Hi @bbulkow, thanks again for a great port and for keeping this port maintained with the latest esp32 fastled improvements. With dual mode bluetooth controller enabled **on esp-idf v4.2**, trying to run your `fastfade` on a single APA102 status led throws a LoadProhibited exception if my data pin is GPIO 19. The animation appears to run fine on WS2812B leds using GPIO 19 for data out.

I basically just cloned the master branch, enabled bluetooth and flashed the app. I have no bluetooth-specific code, just the component enabled and **set to dual mode**.

Steps to reproduce:
- Clone fresh master branch
- Copy sdkconfig.4-2 over to sdkconfig
- In menuconfig, enable dual mode bluetooth
- Comment out WS2812B `addLeds` call and switch with the APA102 example you already had in there, but with GPIO 19 as data.
```
FastLED.addLeds<APA102, 19, 15>(leds, NUM_LEDS);
```

Other than this, there are no changes to the main.cpp file. Immediately after "entering app main, call add leds" is logged, the error below is thrown (full application log pasted for context). Is there something about GPIO 19 that I may not understand? I see that it's used as an SPI pin according to [this](https://randomnerdtutorials.com/esp32-pinout-reference-gpios/) article, but [the official datasheet](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf) doesn't seem to suggest that GPIO 19 is reserved when BT is in use.

Thanks!

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7412
load:0x40078000,len:14724
ho 0 tail 12 room 4
load:0x40080400,len:4232
0x40080400: _init at ??:?

entry 0x40080690
I (30) boot: ESP-IDF v4.2-dev-2084-g98d5b5dfd 2nd stage bootloader
I (30) boot: compile time 07:36:10
I (31) boot: chip revision: 1
I (35) boot.esp32: SPI Speed      : 40MHz
I (39) boot.esp32: SPI Mode       : DIO
I (44) boot.esp32: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x063d8 ( 25560) map
I (109) esp_image: segment 1: paddr=0x00016400 vaddr=0x3ffbdb60 size=0x02404 (  9220) load
I (113) esp_image: segment 2: paddr=0x0001880c vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /Users/drewandre/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (119) esp_image: segment 3: paddr=0x00018c18 vaddr=0x40080404 size=0x07400 ( 29696) load
I (139) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x156fc ( 87804) map
0x400d0020: _stext at ??:?

I (170) esp_image: segment 5: paddr=0x00035724 vaddr=0x40087804 size=0x02978 ( 10616) load
0x40087804: multi_heap_assert at /Users/drewandre/esp/esp-idf/components/heap/multi_heap_platform.h:65
 (inlined by) merge_adjacent at /Users/drewandre/esp/esp-idf/components/heap/multi_heap.c:242

I (180) boot: Loaded app from partition at offset 0x10000
I (180) boot: Disabling RNG early entropy source...
I (182) cpu_start: Pro cpu up.
I (186) cpu_start: Application information:
I (191) cpu_start: Project name:     FastLED-idf
I (196) cpu_start: App version:      1
I (200) cpu_start: Compile time:     Sep  2 2020 07:36:07
I (206) cpu_start: ELF file SHA256:  00e68454ee48f8ba...
I (212) cpu_start: ESP-IDF:          v4.2-dev-2084-g98d5b5dfd
I (219) cpu_start: Starting app cpu, entry point is 0x4008174c
0x4008174c: call_start_cpu1 at /Users/drewandre/esp/esp-idf/components/esp32/cpu_start.c:282

I (0) cpu_start: App cpu up.
I (229) heap_init: Initializing. RAM available for dynamic allocation:
I (236) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (242) heap_init: At 3FFC0898 len 0001F768 (125 KiB): DRAM
I (248) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (255) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (261) heap_init: At 4008A17C len 00015E84 (87 KiB): IRAM
I (267) cpu_start: Pro cpu start user code
I (285) spi_flash: detected chip: gd
I (286) spi_flash: flash io: dio
W (286) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (296) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
 entering app main, call add leds
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40082af0  PS      : 0x00060630  A0      : 0x800d2f60  A1      : 0x3ffb1eb0
0x40082af0: __pinMode at /Users/drewandre/Downloads/FastLED-idf/build/../components/FastLED-idf/hal/esp32-hal-gpio.c:131

A2      : 0x00000013  A3      : 0x00000002  A4      : 0x00000000  A5      : 0x3ffb3b50
A6      : 0x00000000  A7      : 0x3ffb3b60  A8      : 0x3f404860  A9      : 0x00000000
A10     : 0x00000074  A11     : 0x0a732520  A12     : 0x00060620  A13     : 0x00060623
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0a732520  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff7

Backtrace:0x40082aed:0x3ffb1eb0 0x400d2f5d:0x3ffb1ed0 0x400d37c9:0x3ffb1ef0 0x400d3186:0x3ffb1f10 0x400d1907:0x3ffb1f40
0x40082aed: __pinMode at /Users/drewandre/Downloads/FastLED-idf/build/../components/FastLED-idf/hal/esp32-hal-gpio.c:131

0x400d2f5d: APA102Controller<(unsigned char)19, (unsigned char)15, (EOrder)10, 20u>::init() at /Users/drewandre/Downloads/FastLED-idf/build/../components/FastLED-idf/platforms/esp/32/fastpin_esp32.h:27
 (inlined by) ?? at /Users/drewandre/Downloads/FastLED-idf/build/../components/FastLED-idf/fastspi_bitbang.h:39
 (inlined by) APA102Controller<(unsigned char)19, (unsigned char)15, (EOrder)10, 20u>::init() at /Users/drewandre/Downloads/FastLED-idf/build/../components/FastLED-idf/chipsets.h:231

0x400d37c9: CFastLED::addLeds(CLEDController*, CRGB*, int, int) at /Users/drewandre/Downloads/FastLED-idf/build/../components/FastLED-idf/FastLED.cpp:39 (discriminator 4)

0x400d3186: app_main at /Users/drewandre/Downloads/FastLED-idf/build/../components/FastLED-idf/FastLED.h:252
 (inlined by) app_main at /Users/drewandre/Downloads/FastLED-idf/build/../main/main.cpp:272

0x400d1907: main_task at /Users/drewandre/esp/esp-idf/components/esp32/cpu_start.c:589

ELF file SHA256: 00e68454ee48f8ba

Rebooting...
```

bbulkow, 2020-09-07:
Did you have the same issue with your environment with a WS2811 format string, and with bluetooth disabled?

Thanks for the report, I'll give it a try.

bbulkow, 2020-09-08:
Oh, please notice, you're using the 4.2 SKD with master, which is post 4.2. Don't do that.

When using master, you should generate your own SDKCONFIG based on the master you have. Basically, remove sdkconfig, run menuconfig and set for your chip, and run. Espressif's SDKCONFIG is simply not very stable, it seems they are always adding and removing things. 

There's nothing specific in the FastLED config that requires sdkconfig settings (afaik so far :-). I will update the doc to make that more clear.

drewandre, 2020-09-08:
Hi @bbulkow -- when you suggest trying WS2811 with bluetooth disabled, you're basically asking if the example runs after a fresh clone, right? Because the `LED_TYPE` is already WS2811 and bluetooth is disabled in the example on the master branch. But to answer your question, yeah it runs out of the box.

I re-generated the sdkconfig from scratch like you suggested. After enabling bluetooth (not even in dual-mode like I previously mentioned, just all default bluetooth configurations), setting the LED type to APA102 and settings it's data pin to 19 (with a clock pin of 15), I get the same error as above even with the (awesome) changes you pushed to master four days ago to address flickering.

bbulkow, 2020-09-09:
Sorry if I was unclear. I was asking if you could try your test case using a fresh sdkconfig. In your description, you've mismatched using the "sdkconfig4.2" with a master tree. Can you do exactly what you did, but instead of using one of the checked in sdkconfig, delete the sdkconfig completely, do an idf.py menuconfig to create a new one, set whatever you want to set, and try again?

drewandre, 2020-09-11:
Hey @bbulkow, in my comment above I mentioned that I did generate the sdkconfig from scratch like you suggested. I totally blew away all sdkconfig files and re-ran menuconfig to generate a new one. Let me know if I can help debug at all!

bbulkow, 2020-09-12:
Good news is it's reproducing on my machine. Should have an update shortly.

bbulkow, 2020-09-12:
Going to mark this as resolved. Please reopen if problem has not been resolved with today's checkins.

closed 2020-09-12 by bbulkow