bbulkow/FastLED-idf #15: Crash during show

legonigel, 2020-09-09:
Ran into this with esp-idf 4.1 trying to run the provided main.

My investigation shows that it gets to `.show()`. I tried a ton of things and eventually found that setting `FASTLED_RMT_BUILTIN_DRIVER` to `1` made the issue disappear.  I don't fully understand the implications of this, and if I should just live with it set to `1`, but I don't need to use the RMT for anything else and the docs seems to make it seem like this isn't a great idea. 

I am running this on a tinypico, if hardware or chip would make a difference. 

Let me know if I can provide any other info.

Log here, although I don't think they'll be of any use. I could reproduce this with slightly different stack traces by changing compile options and the stack trace doesn't seem to have anything to do with the actual function. 
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7192
load:0x40078000,len:14036
load:0x40080400,len:4152
0x40080400: _init at ??:?

entry 0x40080690
I (29) boot: ESP-IDF v4.1 2nd stage bootloader
I (29) boot: compile time 01:05:26
I (30) boot: chip revision: 1
I (32) boot.esp32: SPI Speed      : 40MHz
I (37) boot.esp32: SPI Mode       : DIO
I (42) boot.esp32: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x06224 ( 25124) map
I (107) esp_image: segment 1: paddr=0x0001624c vaddr=0x3ffb0000 size=0x0219c (  8604) load
I (110) esp_image: segment 2: paddr=0x000183f0 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/nigel/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (116) esp_image: segment 3: paddr=0x000187fc vaddr=0x40080404 size=0x0781c ( 30748) load
I (137) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x15fec ( 90092) map
0x400d0020: _stext at ??:?

I (169) esp_image: segment 5: paddr=0x00036014 vaddr=0x40087c20 size=0x025e4 (  9700) load
0x40087c20: rtc_clk_32k_enable_common at /home/nigel/esp/esp-idf/components/soc/esp32/rtc_clk.c:118

I (179) boot: Loaded app from partition at offset 0x10000
I (179) boot: Disabling RNG early entropy source...
I (181) cpu_start: Pro cpu up.
I (185) cpu_start: Application information:
I (189) cpu_start: Project name:     FastLED-idf
I (195) cpu_start: App version:      83c0736-dirty
I (200) cpu_start: Compile time:     Sep  9 2020 01:05:23
I (206) cpu_start: ELF file SHA256:  001c818c895b989b...
I (212) cpu_start: ESP-IDF:          v4.1
I (217) cpu_start: Starting app cpu, entry point is 0x40081160
0x40081160: call_start_cpu0 at /home/nigel/esp/esp-idf/components/esp32/cpu_start.c:205 (discriminator 1)

I (209) cpu_start: App cpu up.
I (228) heap_init: Initializing. RAM available for dynamic allocation:
I (234) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (240) heap_init: At 3FFB2BF0 len 0002D410 (181 KiB): DRAM
I (247) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (253) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (259) heap_init: At 4008A204 len 00015DFC (87 KiB): IRAM
I (266) cpu_start: Pro cpu start user code
I (284) spi_flash: detected chip: gd
I (284) spi_flash: flash io: dio
I (284) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
 entering app main, call add leds
 set max power
create task for led blinking
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d55d4  PS      : 0x00060033  A0      : 0x800d48e5  A1      : 0x3ffaf8b0
0x400d55d4: fopen at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fopen.c:168

A2      : 0x00000000  A3      : 0x3ffb1390  A4      : 0x00000040  A5      : 0x3ffb1390
A6      : 0x000000b0  A7      : 0x00000000  A8      : 0x00000034  A9      : 0x00000000
A10     : 0x00000001  A11     : 0x00009470  A12     : 0x3ff560b0  A13     : 0xefffffff
A14     : 0xfff7ffff  A15     : 0xffffffbf  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400d4700  LEND    : 0x400d4711  LCOUNT  : 0x00000000
0x400d4700: rtc_isr at /home/nigel/esp/esp-idf/components/driver/rtc_module.c:72

0x400d4711: rtc_isr at /home/nigel/esp/esp-idf/components/driver/rtc_module.c:76

ELF file SHA256: 001c818c895b989b

Backtrace: 0x400d55d1:0x3ffaf8b0 0x400d48e2:0x3ffaf8e0 0x40082aec:0x3ffaf920 0x400d4257:0x3ffaf940 0x400d4649:0x3ffaf9a0 0x400d3b5e:0x3ffaf9e0 0x400d1e5c:0x3ffafa10
0x400d55d1: fopen at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fopen.c:167

0x400d48e2: uart_set_stop_bits at /home/nigel/esp/esp-idf/components/driver/uart.c:202 (discriminator 5)

0x40082aec: spi_flash_cache_enabled at /home/nigel/esp/esp-idf/components/spi_flash/cache_utils.c:340

0x400d4257: esp_efuse_utility_process at /home/nigel/esp/esp-idf/components/efuse/src/esp_efuse_utility.c:63

0x400d4649: periph_module_enable at /home/nigel/esp/esp-idf/components/driver/periph_ctrl.c:33 (discriminator 1)

0x400d3b5e: esp_flash_init_default_chip at /home/nigel/esp/esp-idf/components/spi_flash/esp_flash_spi_init.c:183

0x400d1e5c: esp_ipc_call_and_wait at /home/nigel/esp/esp-idf/components/esp_common/src/ipc.c:115

Rebooting...
```

legonigel, 2020-09-10:
I can now confirm that this does _not_ happen on 4.2, so this seems to be a 4.1 issue.

Thenicolaibulow, 2020-09-10:
I can repport a very similar issue, though with version 4.2. 
I'm (trying to) run 512 WS2812B's on a ESP32 Wroom, and am experiencing a very similar hard crash, whenever .show() i called.

`
entry 0x40080690
I (28) boot: ESP-IDF v4.2-dev-459-ge36516372-dirty 2nd stage bootloader
I (28) boot: compile time 12:28:27
I (29) boot: chip revision: 1
I (33) boot.esp32: SPI Speed      : 40MHz
I (38) boot.esp32: SPI Mode       : DIO
I (42) boot.esp32: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0633c ( 25404) map
I (108) esp_image: segment 1: paddr=0x00016364 vaddr=0x3ffb0000 size=0x02238 (  8760) load
I (111) esp_image: segment 2: paddr=0x000185a4 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /Users/nicolaidyre/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1817

I (117) esp_image: segment 3: paddr=0x000189b0 vaddr=0x40080404 size=0x07668 ( 30312) load
I (138) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x15e88 ( 89736) map
0x400d0020: _stext at ??:?

I (170) esp_image: segment 5: paddr=0x00035eb0 vaddr=0x40087a6c size=0x025f4 (  9716) load
0x40087a6c: vPortExitCritical at /Users/nicolaidyre/esp/esp-idf/components/freertos/xtensa/port.c:410

I (179) boot: Loaded app from partition at offset 0x10000
I (179) boot: Disabling RNG early entropy source...
I (181) cpu_start: Pro cpu up.
I (185) cpu_start: Application information:
I (190) cpu_start: Project name:     FastLED-idf
I (195) cpu_start: App version:      83c0736-dirty
I (201) cpu_start: Compile time:     Sep 10 2020 12:28:33
I (207) cpu_start: ELF file SHA256:  04fc65b207ca56f2...
I (213) cpu_start: ESP-IDF:          v4.2-dev-459-ge36516372-dirty
I (220) cpu_start: Starting app cpu, entry point is 0x40081164
0x40081164: call_start_cpu1 at /Users/nicolaidyre/esp/esp-idf/components/esp32/cpu_start.c:274

I (0) cpu_start: App cpu up.
I (230) heap_init: Initializing. RAM available for dynamic allocation:
I (237) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (243) heap_init: At 3FFB2F00 len 0002D100 (180 KiB): DRAM
I (249) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (255) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (262) heap_init: At 4008A060 len 00015FA0 (87 KiB): IRAM
I (268) cpu_start: Pro cpu start user code
I (286) spi_flash: detected chip: gd
I (287) spi_flash: flash io: dio
W (287) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (297) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
 entering app main, call add leds
 set max power
create task for led blinking
blink leds
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d5908  PS      : 0x00060f33  A0      : 0x800d4671  A1      : 0x3ffb7b50  
0x400d5908: rmt_set_tx_thr_intr_en at /Users/nicolaidyre/esp/esp-idf/components/driver/rmt.c:389

A2      : 0x00000000  A3      : 0x3ffb1394  A4      : 0x00000040  A5      : 0x3ffb1394  
A6      : 0x000000b0  A7      : 0x00000000  A8      : 0x00000034  A9      : 0x00000000  
A10     : 0x3ffb236c  A11     : 0x00009470  A12     : 0x3ff560b0  A13     : 0xefffffff  
A14     : 0xfff7ffff  A15     : 0xffffffbf  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400d4478  LEND    : 0x400d4489  LCOUNT  : 0x00000000  
0x400d4478: calculate_unscaled_power_mW(CRGB const*, unsigned short) at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/power_mgt.cpp:60

0x400d4489: calculate_unscaled_power_mW(CRGB const*, unsigned short) at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/power_mgt.cpp:59

ELF file SHA256: 04fc65b207ca56f24f06871c87e8d00d8337f96ff2148385ae6e888246b89aa5

Backtrace: 0x400d5905:0x3ffb7b50 0x400d466e:0x3ffb7b80 0x40082814:0x3ffb7bc0 0x400d4003:0x3ffb7be0 0x400d43f1:0x3ffb7c40 0x400d3945:0x3ffb7c80
0x400d5905: rmt_ll_set_tx_limit at /Users/nicolaidyre/esp/esp-idf/components/soc/src/esp32/include/hal/rmt_ll.h:177
 (inlined by) rmt_set_tx_thr_intr_en at /Users/nicolaidyre/esp/esp-idf/components/driver/rmt.c:389

0x400d466e: ESP32RMTController::init() at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/platforms/esp/32/clockless_rmt_esp32.cpp:233

0x40082814: ESP32RMTController::showPixels() at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/platforms/esp/32/clockless_rmt_esp32.cpp:265

0x400d4003: CPixelLEDController<(EOrder)10, 1, 4294967295u>::show(CRGB const*, int, CRGB) at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/platforms/esp/32/clockless_rmt_esp32.h:394
 (inlined by) CPixelLEDController<(EOrder)10, 1, 4294967295u>::show(CRGB const*, int, CRGB) at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/controller.h:408

0x400d43f1: CFastLED::show(unsigned char) at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/controller.h:90
 (inlined by) CFastLED::show(unsigned char) at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/FastLED.cpp:59

0x400d3945: blinkLeds_simple(void*) at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/FastLED.h:506
 (inlined by) blinkLeds_simple(void*) at /Users/nicolaidyre/esp/FastLED-idf/build/../main/main.cpp:166

Rebooting...
`

bbulkow, 2020-09-12:
@legonigel  Can you say what number of LEDs you have, what model, what pins?
I have resolved issues regarding how RTC pins were locked today, but I don't think this solves your issue.
It sounds more like a blown stack based on a large number of LEDs. Please let me know your led count so I can re-test.

legonigel, 2020-09-12:
> @legonigel Can you say what number of LEDs you have, what model, what pins?
> I have resolved issues regarding how RTC pins were locked today, which might help your issue, but those were crashing in a very different place, and not on the RMT code path.

I was running 5 WS2812b LEDs, So I don't think this was a number of LEDs Issue. I tested on GPIO26 and GPIO27. I'm only trying to run a single strip of 36 LEDs, so nothing crazy. 

> I am actually deeply shocked that the built-in driver works, will take a moment to re-test that also... I had sorta given up on it.

I never actually got it to run the strip, so I wouldn't say it worked, but it didn't crash the ESP. It would output like 2.75 LEDs worth do data (I forget the exact number of bits, I think ~70 bits) then stop. I was outputting some really low frequency data, then a couple times per second a burst of correct data. I never bothered to try to get this working. 

> I've also been testing primarily on 4.0 ( because it's "stable" ), 4.2 ( because it supports the S2 ), and master, because it's master :-)

I'm happy to just run 4.2 for my testing. I believe 4.1 is now considered "stable", which is why I was originally running it.

bbulkow, 2020-09-13:
I wanted to make sure there wasn't a hidden stack allocation on a path somewhere, and didn't see anything crazy, which is what I was worried about. I've got 4.1 cloning and will see if I can get it to crash, try a lot of LEDs on both, etc.

bbulkow, 2020-09-14:
@Thenicolaibulow  your issue must be different. I'll test 512 on 4.2 only, and may break out your issue, unless you want to create a new one yourself. @legonigel  's issue is clearly 4.1 specific.

Thenicolaibulow, 2020-09-14:
Thanks @bbulkow I'd appreciate that. I can of course not outrule, that i might be facing bigger issues than mentioned here, as i've had problems with other libraries too. With other libraries i've been able to get the LED's to run, however it's been with quite a lot of glitching after 450th LED, which i've been suspecting is due to the "interrupt issue" which, from i can tell, is exactly what you're trying to mitigate with the RMT-based implementation. 
Anyways, let me know what you find. 
I'll happily post another more detailed issue, if that helps!

Thanks, 
\n

bbulkow, 2020-09-15:
Resolved issue with ESP-IDF 4.1, required a bit of work. Please let me know if you have further problems.

closed 2020-09-15 by bbulkow