bbulkow/FastLED-idf #16: More than one channel doesn't work with mem_blocks_num

bbulkow, 2020-09-14:
If you use the default code which uses mem_blocks_num of 2 ( or increase it further ), ESP-IDF demands that you skip channels. For example, if you use channel 0 with 2 memblocks, then you must skip channel 1 because its memory is already in use.

Until this issue is resolved, you can use one LED strip with greater mem_blocks, or you can use a mem_blocks_num of 1 and multiple strips. Looking at a fix over the next few days, pull requests welcome.

Thanks @samguyer for bringing this up.

samguyer, 2020-09-14:
@bbulkow  I just add add the number of mem blocks to the channel as I'm assigning them:

        int channel = 0;
        while (channel < FASTLED_RMT_MAX_CHANNELS && gNext < gNumControllers) {
            ESP32RMTController::startNext(channel);
            // -- Important: when we use more than one memory block, we need to
            //    skip the channels that would otherwise overlap in memory.
            channel += FASTLED_RMT_MEM_BLOCKS;
        }

bbulkow, 2020-09-14:
Sweet and thanks. It's probably time for me to test gNumControllers > RMT_MAX_CHANNELS, which will happen more frequently when RMT_MEM_BLOCKS > 1, want to make sure all that works even with this extra complication.

samguyer, 2020-09-14:
I was thinking the same thing.I think it will work, since the next controller just takes the channel number that was freed up.

bbulkow, 2020-09-14:
Checked in, will test this afternoon.... couldn't get any worse.... :-0

bbulkow, 2020-09-15:
Resolved, hopefully.

samguyer, 2020-09-15:
I just tell people that if they're not happy, I'll give them a full refund. ;-)

bbulkow, 2020-09-15:
Multiple channels crashes immediately. Need to fix.

samguyer, 2020-09-15:
Ok, I'll take a look tomorrow.

bbulkow, 2020-09-15:
Not in your version. I wrote it a different way.

bbulkow, 2020-09-15:
Resolved with ba32cc3

closed 2020-09-15 by bbulkow