bbulkow/FastLED-idf #17: Crash with 512 LEDs

bbulkow, 2020-09-15:
I can report a very similar issue, though with version 4.2.
I'm (trying to) run 512 WS2812B's on a ESP32 Wroom, and am experiencing a very similar hard crash, whenever .show() i called.

`
entry 0x40080690
I (28) boot: ESP-IDF v4.2-dev-459-ge36516372-dirty 2nd stage bootloader
I (28) boot: compile time 12:28:27
I (29) boot: chip revision: 1
I (33) boot.esp32: SPI Speed : 40MHz
I (38) boot.esp32: SPI Mode : DIO
I (42) boot.esp32: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label Usage Type ST Offset Length
I (63) boot: 0 nvs WiFi data 01 02 00009000 00006000
I (71) boot: 1 phy_init RF data 01 01 0000f000 00001000
I (78) boot: 2 factory factory app 00 00 00010000 00100000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0633c ( 25404) map
I (108) esp_image: segment 1: paddr=0x00016364 vaddr=0x3ffb0000 size=0x02238 ( 8760) load
I (111) esp_image: segment 2: paddr=0x000185a4 vaddr=0x40080000 size=0x00404 ( 1028) load
0x40080000: _WindowOverflow4 at /Users/nicolaidyre/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1817

I (117) esp_image: segment 3: paddr=0x000189b0 vaddr=0x40080404 size=0x07668 ( 30312) load
I (138) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x15e88 ( 89736) map
0x400d0020: _stext at ??:?

I (170) esp_image: segment 5: paddr=0x00035eb0 vaddr=0x40087a6c size=0x025f4 ( 9716) load
0x40087a6c: vPortExitCritical at /Users/nicolaidyre/esp/esp-idf/components/freertos/xtensa/port.c:410

I (179) boot: Loaded app from partition at offset 0x10000
I (179) boot: Disabling RNG early entropy source...
I (181) cpu_start: Pro cpu up.
I (185) cpu_start: Application information:
I (190) cpu_start: Project name: FastLED-idf
I (195) cpu_start: App version: 83c0736-dirty
I (201) cpu_start: Compile time: Sep 10 2020 12:28:33
I (207) cpu_start: ELF file SHA256: 04fc65b207ca56f2...
I (213) cpu_start: ESP-IDF: v4.2-dev-459-ge36516372-dirty
I (220) cpu_start: Starting app cpu, entry point is 0x40081164
0x40081164: call_start_cpu1 at /Users/nicolaidyre/esp/esp-idf/components/esp32/cpu_start.c:274

I (0) cpu_start: App cpu up.
I (230) heap_init: Initializing. RAM available for dynamic allocation:
I (237) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (243) heap_init: At 3FFB2F00 len 0002D100 (180 KiB): DRAM
I (249) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (255) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (262) heap_init: At 4008A060 len 00015FA0 (87 KiB): IRAM
I (268) cpu_start: Pro cpu start user code
I (286) spi_flash: detected chip: gd
I (287) spi_flash: flash io: dio
W (287) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (297) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
entering app main, call add leds
set max power
create task for led blinking
blink leds
Guru Meditation Error: Core 0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC : 0x400d5908 PS : 0x00060f33 A0 : 0x800d4671 A1 : 0x3ffb7b50
0x400d5908: rmt_set_tx_thr_intr_en at /Users/nicolaidyre/esp/esp-idf/components/driver/rmt.c:389

A2 : 0x00000000 A3 : 0x3ffb1394 A4 : 0x00000040 A5 : 0x3ffb1394
A6 : 0x000000b0 A7 : 0x00000000 A8 : 0x00000034 A9 : 0x00000000
A10 : 0x3ffb236c A11 : 0x00009470 A12 : 0x3ff560b0 A13 : 0xefffffff
A14 : 0xfff7ffff A15 : 0xffffffbf SAR : 0x00000020 EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000 LBEG : 0x400d4478 LEND : 0x400d4489 LCOUNT : 0x00000000
0x400d4478: calculate_unscaled_power_mW(CRGB const*, unsigned short) at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/power_mgt.cpp:60

0x400d4489: calculate_unscaled_power_mW(CRGB const*, unsigned short) at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/power_mgt.cpp:59

ELF file SHA256: 04fc65b207ca56f24f06871c87e8d00d8337f96ff2148385ae6e888246b89aa5

Backtrace: 0x400d5905:0x3ffb7b50 0x400d466e:0x3ffb7b80 0x40082814:0x3ffb7bc0 0x400d4003:0x3ffb7be0 0x400d43f1:0x3ffb7c40 0x400d3945:0x3ffb7c80
0x400d5905: rmt_ll_set_tx_limit at /Users/nicolaidyre/esp/esp-idf/components/soc/src/esp32/include/hal/rmt_ll.h:177
(inlined by) rmt_set_tx_thr_intr_en at /Users/nicolaidyre/esp/esp-idf/components/driver/rmt.c:389

0x400d466e: ESP32RMTController::init() at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/platforms/esp/32/clockless_rmt_esp32.cpp:233

0x40082814: ESP32RMTController::showPixels() at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/platforms/esp/32/clockless_rmt_esp32.cpp:265

0x400d4003: CPixelLEDController<(EOrder)10, 1, 4294967295u>::show(CRGB const*, int, CRGB) at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/platforms/esp/32/clockless_rmt_esp32.h:394
(inlined by) CPixelLEDController<(EOrder)10, 1, 4294967295u>::show(CRGB const*, int, CRGB) at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/controller.h:408

0x400d43f1: CFastLED::show(unsigned char) at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/controller.h:90
(inlined by) CFastLED::show(unsigned char) at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/FastLED.cpp:59

0x400d3945: blinkLeds_simple(void*) at /Users/nicolaidyre/esp/FastLED-idf/build/../components/FastLED-idf/FastLED.h:506
(inlined by) blinkLeds_simple(void*) at /Users/nicolaidyre/esp/FastLED-idf/build/../main/main.cpp:166

Rebooting...
`

bbulkow, 2020-09-15:
Reported by @Thenicolaibulow 

Thanks @bbulkow I'd appreciate that. I can of course not outrule, that i might be facing bigger issues than mentioned here, as i've had problems with other libraries too. With other libraries i've been able to get the LED's to run, however it's been with quite a lot of glitching after 450th LED, which i've been suspecting is due to the "interrupt issue" which, from i can tell, is exactly what you're trying to mitigate with the RMT-based implementation.
Anyways, let me know what you find.

I'll happily post another more detailed issue, if that helps!

Thanks,
\n

bbulkow, 2020-09-15:
I just tried simply setting the test code to 1024 LEDs and the code was stable. There have been quite a few changes since you last tried, can you try again? 

And also mention the structure of your leds... all one controller? or many pins?

Thenicolaibulow, 2020-09-15:
Hi! @bbulkow 

I replaced my IDF with v4.1, and it all seems to be working now. Thanks!
Now on to tackling my power supply solution.. (currently running on a LM2596 which is just HORRIBLE!)

Thanks for all the work. Really appreciate it!

\n

closed 2020-09-15 by bbulkow