bbulkow/FastLED-idf #18: Restructure repository for the components

outlandnish, 2020-09-15:
To use this FastLED-idf component in my code and keep up to date with your changes, I keep my own respository that tracks these changes as an IDF component here: https://github.com/intentfulmotion/FastLED-idf

This allows me to use it as a submodule in my project and not have to keep a version of this component's source in my project repository (https://github.com/intentfulmotion/firmware-amp). It's the same repository structure used by Espressif to maintain their different components. 

Would you be open to breaking off the FastLED-idf and WS2812FX-idf components into their own repositories and then using them as submodules in this repository similarly?

bbulkow, 2020-09-27:
Yes, that's a good idea.

BorisKourt, 2020-10-08:
I agree with this. Structurally it might be better to move the examples 'wrapping' out of this repository, and just leave the FastLED component here. That way the bulk of the work remains here, historical changes are clearly preserved, and the examples become optional. 

Essentially:

```
bbulkow/FastLED-idf 
  hal/
  lib8tion/
  platforms/
  ...

bbulkov/WS2812FX-idf
  CMakeLists.txt
  FX.cpp
  ...

bbulkov/FastLED-idf-Examples
  components/FastLED-idf(submodule)
  components/WS2812FX-idf(submodule)
  main/
  ...
```

I'd gladly PR a proposal, but sadly can't just create repos under your name! :D

n3mawashi, 2023-12-02:
Think notionally this is a good idea, but examples should stay with the respective repositories to make it easier for others to pickup and use.