bbulkow/FastLED-idf #20: Glitching issues with patterns

ntwallace, 2020-10-06:
First off, thanks for this port--this was the push I needed to migrate my big LED projects from Arduino to IDF with Arduino as a component. Let me know if I can be of help with the hardware SPI driver at all.

I'm testing out running WS2812FX patterns on WS2812B LEDs and noticing a few different glitching and ghosting issues:

1. The last pixel always glitches white when I run a pattern. This happens on all patterns and with both the RMT and I2S driver. Disabling WiFi and Bluetooth doesn't make a difference. Here's an example running a color wipe on a 256 pixel matrix, same thing happens on standard strips: [GIF link](https://media4.giphy.com/media/b9ovMhZpo16jGjmx1X/giphy.gif)

2. I think there may be a data overflow with the I2S driver while running patterns. When using the I2S driver, I get a bunch of extra glitching pixels on past the amount I have defined for the controller. I don't have this issue with the RMT driver. This is the same setup as above (256 LEDs on one output), when I attach additional LEDs I get all this extra glitchy stuff: [GIF link](https://media3.giphy.com/media/8DlbdMvoN6vTEYQTbg/giphy.gif)

3. I see some ghosting on the leading pixels of animations when WiFi or Bluetooth is enabled. It's hard to capture with my phone camera, but in the above two GIFs there's also an issue where the leading few pixels of the animation don't clear/animate smoothly sometimes. Disabling both WiFi and Bluetooth fixes the issue, so I assume this is due to jitter. I noticed in the I2S driver you put fillBuffer in IRAM, but I don't think the 0 flag for the esp_inter_alloc call adds `ESP_INTR_FLAG_IRAM`, does it? I can explicitly call `ESP_INTR_FLAG_LEVEL` as 1, 2, or 3, but if I add `| ESP_INTR_FLAG_IRAM` I no longer get any output. Setting `ESP_INTR_FLAG_LEVEL` to 4 or greater also results in no output. Any thoughts on how to set the priority higher to avoid jitter issues?

Thanks again!

BorisKourt, 2020-10-08:
I am also having these issues. The leading pixels are ghosting 'ahead' by a semi-random number. 
The board I am using is Adafruit's Huzzah32. Data pin is set to 33 in the code, I haven't been able to find information on whether this pin is problematic or not.

I need WiFi for my application, and disabling just bluetooth on its own doesn't have an effect.

Since there is additional code running along side the FastLED animation, is there something that I should do to help isolate it from the rest?

Can I do anything to provide more details for the issue?

Thanks for any advice!

bbulkow, 2020-10-17:
@ntwallace  You can't set the FLAG_LEVEL to 4 or greater because the ESP-IDF framework doesn't allow C programs to run at those levels. This is documented https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/hlinterrupts.html .

1. "last pixel always glitches" - given that all my test patterns on I2S and RMT work _except for a particular I2S issue_, I would find it hard to understand that there's a code problem causing the last pixel that's caused by FastLED given how the code's so different between I2S and RMT.

2. Yes, we've now got 4 reports of this. What is necessary is not a picture but simple sample code that does it. I spent an afternoon writing test patterns that didn't trigger it.

3. Yes, first pixel flashes are usually jitter issues. The RMT driver has had lots of focus to remove jitter, and the I2S code hasn't. In the cases where you "don't see output" by raising the interrupt level, that's almost certainly because registering the interrupt has returned an error code, and Yve's code ignores errors. I'll put the error wrapper on that one function because it's a good one. 

I don't have a lot of energy myself in fixing potential I2S jitter problems because I don't have them, but you can work with Yves, and I'll be happy to upstream his code fixes or yours.

In order to improve I2S with jitter, it'll be necessary to determine if the problem is Wifi or Bluetooth, to see if there's a way to increase the buffering of the I2S code (like we did with RMT), whether the core you run on matters, whether you can lower the priority of wifi and/or bluetooth, and finally whether there's a section of code that can be changed to ASM in order to allow higher level interrupt access. Maybe there are other areas to explore but that makes sense to me.

bbulkow, 2020-10-17:
@BorisKourt . I don't know what you could mean by "ghosting ahead". Ghosting is when you turn something (like a pixel) off, and it stays lit for a while, and then goes off ( like a ghost ).

If you think you have a jitter problem, I believe the readme.md suggests that you switch from I2S to RMT, and then you turn on the "SHOWPIXELS" print so you can see if there are any "bailed" messages. It also allows you to see the number of microseconds between interrupts, and the amount of jitter, thus allowing you to raise the RMT buffer sizes to overcome.

The only code issue I know about right now which I can work on is the issue with I2S generating the wrong patterns - simply wrong. I have written some test code which doesn't show this, and I have some complex test code that might show the pattern, and one submission from a user that they claim does show the I2S pattern incorrect problem. If you can submit (simple) code that shows an I2S incorrect pattern ( which works fine on RMT ) that would be a great help.

ShubhamS156, 2021-12-01:
Hey,
I'm having a weird problem that whenever I restart my ESP32 the leds show a different color+pattern even if the code just tells it to set a static color.

psy0rz, 2023-01-16:
For anyone having glitching issues while using wifi: Try changing the core affinity: I configured "Main task core affinity (CPU1)" and all other stuff i could find to CPU0.

I only have a app_main() and i'm not creating any tasks.

No glitching while streaming data via wifi and spamming serial. :)