bbulkow/FastLED-idf #21: Unable to change I2S_DEVICE from 0 to 1

drewandre, 2020-10-15:
Hi @bbulkow, thanks again for this port and for the quality of maintenance. I've found it a breeze to work with so far and the readme is above and beyond.

I'm using this port in a project that uses the SGTL5000 audio codec on I2S bus 0 (which this port uses by default). BT audio stream comes in and is handled by I2S bus 0, pumped through a filter bank for audio analysis, and sent to the SGTL5000 using GPIO 0 for main clock.

I see that you should be able to change the default I2S bus in [platforms/esp/32/clockless_i2s_esp32.h line 127](https://github.com/bbulkow/FastLED-idf/blob/master/components/FastLED-idf/platforms/esp/32/clockless_i2s_esp32.h#L127), but when I set this to 1, nothing shows on the WS2811 strip I'm using. I understand i2s at a high level, but I'm not sure how to debug this, nor do I have access to a scope.

Is there something I'm missing here about which pins are usable with each i2s bus? I ask because I know i2s bus 0 must use GPIO 0 for its main clock, so maybe I'm just using the wrong pins? (I was running the included sketch on pin 13)

Steps to reproduce:
- Clone latest FastLED-idf
- Copy over the sdkconfig.4-2 config to sdkconfig, run `idf.py menuconfig`, and save
- Do not change any of the pins/led sets used (I own a WS2811 strip and ran it on pin 13)
- Run `idf.py flash monitor` and ensure included `blinkWithFx_test` fxn renders animations as expected
- Add `#define I2S_DEVICE 1` to very top of main.cpp
- Run `idf.py flash monitor` again and observe that none of the animations are displayed

It would be great to have the flexibility to run FastLED using i2s bus 1, and all audio on i2s bus 0. That being said, is there anything I should watch out for here? Theoretically I should be able to do this, right? I remember earlier this summer when I was using this port and its RMT interface, outputting LED data completely killed my audio. It was all scrambled as if the RMT was interrupting the main clock or something. But it was way beyond my experience level and I didn't feel opening an issue was necessary. But now that the i2s interface has been ironed out in this port, I thought maybe this is again a good time to see if this port plays well with my audio pipeline. Happy to share code samples, too. Thanks!

Versions:
- ESP-IDF 4.2 (from release/4.2 branch)
- FastLED-idf (latest)

bbulkow, 2020-10-17:
Thanks, this code comes from Yves, I'll check with him.

drewandre, 2020-10-17:
Thanks, this is the last item before I'm fully able to switch over from arduino-esp32 to a full esp-idf environment. Let me know if I can help at all or share my audio codec configuration.

drewandre, 2020-11-05:
Hi @bbulkow, just wondering if you ever heard back from Yves? Would this issue be better suited on one of his github repos? Let me know if I can help at all -- thanks.

Spencatro, 2021-12-02:
+1 on this one. I think there's gotta be something in `clockless_i2s_esp32.h` hardcoded to 0 somewhere, maybe?

Spencatro, 2021-12-02:
Left a comment on FastLED#1165 as well, but it looks like this may just be a limitation of I2S1: in the [technical reference manual](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf)--page 303, section 12.1, note 1: "... (note that I2S1 does not support 24-bit width)."

Bummer!