bbulkow/FastLED-idf #24: Implementing FastLED-idf with espressif esp-homekit-sdk

camielverdult, 2021-04-01:
Hello,

I am trying to make a firmware (based on example code from [espressif's homekit SDK](https://github.com/espressif/esp-homekit-sdk)) for a board that uses homekit to control WS2812B lights. While trying to add the FastLed component to the [lightbulb](https://github.com/espressif/esp-homekit-sdk/tree/master/examples/lightbulb) example, a ton of warnings and errors occur. I have attached a build log (the output of `idf.py build`) and the modified example. Can you please tell me what I am doing wrong and how I could fix it? 

This is on a MacBook Pro running 11.2.1 with idf version `ESP-IDF v4.4-dev-479-g1067b2870`.

(I have removed the build folder to due github upload size limitation, please ask if I need to provide it)

[build_log.txt](https://github.com/bbulkow/FastLED-idf/files/6244905/build_log.txt)
[ledstrip.zip](https://github.com/bbulkow/FastLED-idf/files/6244907/ledstrip.zip)

drewandre, 2021-04-01:
All files that import FastLED.h need to be .cpp files. Have you made this change? For example my project which uses FastLED-idf + esp-homekit-sdk has an app_main.cpp file, and all other files are also *.cpp files.

camielverdult, 2021-04-01:
The only thing I did was dragging components/FastLED-idf to the components folder of esp-homekit-sdk. I see [here](https://github.com/bbulkow/FastLED-idf#fastled) That you should drop FastLED as well?! Is there an installation/starting out section?

drewandre, 2021-04-01:
You dropped _just_ [the FastLED-idf component](https://github.com/bbulkow/FastLED-idf/tree/master/components/FastLED-idf) into your components folder, right? Make sure you didn't put this entire repository in your components folder because the actual FastLED-idf component is inside this repository's components directory.

Also make sure you clean your project (`idf.py fullclean`)

camielverdult, 2021-04-01:
I tried to replicate errors after every little step. It appears that after changing the app_main and lightbulb c files to be cpp files, it won't compile. This is without including FastLED. Any idea? [Link to example](https://github.com/espressif/esp-homekit-sdk/tree/master/examples/lightbulb)

dandre-hound, 2021-04-01:
What does your CMakeLists.txt look like? Make sure to include these .cpp files. Here is mine:
```
idf_component_register(
  SRCS
    app_main.cpp
    my_animations_file.cpp
    # ... other files

  INCLUDE_DIRS "."
)
```
Also make sure your app_main.cpp has extern "C" like so:
```#ifdef __cplusplus
extern "C"
{
#endif
  void app_main()
  {
    my_code_here();
  }
#ifdef __cplusplus
}
#endif
```

dandre-hound, 2021-04-01:
You may also need to define extern "C" around any files that import FastLED.h

drewandre, 2021-04-01:
What does your CMakeLists.txt look like? Make sure to include these .cpp files. Here is mine:
```
idf_component_register(
  SRCS
    app_main.cpp
    my_animations_file.cpp
    # ... other files

  INCLUDE_DIRS "."
)
```
Also make sure your app_main.cpp has extern "C" like so:
```
#ifdef __cplusplus
extern "C"
{
#endif
  void app_main()
  {
    my_code_here();
  }
#ifdef __cplusplus
}
#endif
```
You may also need to define extern "C" around any files that import FastLED.h

closed 2024-03-27 by camielverdult